jmalins/BattleBot-Control #11: Address bar takes space for Chrome browser on Android

MarianoMunoz, 2017-10-10:
The Address bar, on the mobile Chrome browser for Android, takes up valuable screen space when rotated into horizontal mode. This shifts the on screen touch controls down and makes them difficult to see/access.

jmalins, 2017-10-10:
Based on my research, it looks like the HTML5 Full Screen API is required to fix this on Android (more info [here](https://developers.google.com/web/fundamentals/native-hardware/fullscreen/)).

Seems straightforward to implement using this library:
https://github.com/sindresorhus/screenfull.js

Unfortunately, I don't have an Android device to test with. Can anyone loan me an old phone?

JamesNewton, 2017-10-10:
I have an old MotoDroid 4 with a bad battery, I wouldn't mind putting a new battery in it if it would be of use, but it's up here in Escondido and I won't be down in SD proper any time soon. I believe it's on Android 4.1 so Chrome should just run.
https://en.wikipedia.org/wiki/Droid_4