jmalins/BattleBot-Control #12: Question: Servo vs weaponMotor

Rarefy-Dev, 2017-11-13:
Hello,
I have this up an running, but unsure of the intent on the reverse button and the slider.  I thought that the reverse button ran the weapon motor and the slider ran the  servo, but I'm not seeing reference to the Servo pin 7 in the INO.  is that just not implemented or am i being dense?  Thanks for your time.
Mark

In hardware.json
"weaponMotor": {
      "driver": "PWM",
      "params": {
        "pwmPin": 6,
        "minMicroseconds": 900,
        "maxMicroseconds": 1800
      }
    },
    "servo": {
      "driver": "PWM",
      "params": {
        "pwmPin": 7
      }

jmalins, 2017-12-05:
Hi Mark, sorry for the very belated response. I was in the process of moving to LA and starting a new job, so have been pretty out of it.

The current code is in a relatively unfinished state in certain areas. The forward/reverse buttons are intended to reverse the front and rear of the robot, while the slider controls the `weaponMotor` ESC. This logic is in `robot.js`. 

You are correct that there is no reference to a Servo in the INO yet. The plan is to have the firmware and client both read it and configure things appropriately... currently though, the firmware side is hard-coded to 3 devices, as is the client-side sender. All of the pin and timing definitions in `hardware.json` are ignored, because it's hardcoded in the firmware. Likewise, I defined the servo there so I could test the servo client side objects... but there is no corresponding implementation on the backend, nor is the value sent to the backend. 

Sorry for the confusion and for checking things in in such an inconsistent state. Do you need a servo? If so, I can walk you through adding it the current approach, or we could talk through the longer term details if you want to take a stab at them.

Rarefy-Dev, 2017-12-07:
No worries.  Congrats on the new position.  I put on a free, friends and neighbors battlebot camp over Thanksgiving with your code.  Worked great, kids loved it.  I had no real issue.  I had bought servos and brushed motors for weapons, but we didn't have an enclosure to safely do spinning weapons anyway.    Thank you so much for this tool.

closed 2017-12-07 by Rarefy-Dev