jmalins/BattleBot-Control #13: Compile Error in Arduino IDE for recent code

shinyk78, 2017-11-19:
Hello everyone.
I kindly request some suggestions on what I need to do with the following error I get on compile time. It sees that there are some type mismatch errors going on. 

Arduino: 1.8.5 (Mac OS X), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

/Arduino/libraries/ESPAsyncTCP-master/src/ESPAsyncTCP.cpp: In constructor 'AsyncClient::AsyncClient(tcp_pcb*)':
/Arduino/libraries/ESPAsyncTCP-master/src/ESPAsyncTCP.cpp:78:28: error: invalid conversion from 'long int (*)(void*, tcp_pcb*, pbuf*, long int)' to 'tcp_recv_fn {aka signed char (*)(void*, tcp_pcb*, pbuf*, signed char)}' [-fpermissive]
     tcp_recv(_pcb, &_s_recv);

Thank you all for making all of this available and easily accessible. 
We are excited to join your community of Battlebots.

Rarefy-Dev, 2017-11-19:
Hello shinyk78,
On what step did you see this error?  Did you Install the ESP8266 devtools add-on?  Which usb to serial chip do you have?

Rarefy

JamesNewton, 2017-11-19:
I believe I mentioned this before:
https://github.com/me-no-dev/ESPAsyncTCP/issues/60

shinyk78, 2017-11-20:
INCREDIBLE! Thank you for your response. The new version of the CPP and H files did it. I successfully uploaded project.

closed 2017-11-20 by shinyk78